wendlers/mpfshell #2: not text displayed after prompt (but it is there)

ideal2545, 2016-06-01:
Hi, 

After I launch mpfshell I am presented with a prompt, however I cannot see what I am typing, its hidden almost like a password. I can type in commands and hit enter and they work, but I can't actually see what I am typing... am I missing a library perhaps?

MarkR42, 2016-06-02:
Me too.

I don't get characters echoed in the shell, which makes typing commands challenging.

Tooblippe, 2016-06-06:
Firstly thanks. This is the only way I have been able to move files to the ESP.
It work and looks great. I have the same problem though. Anything I type is not displayed, but the commands are issued and it works.

wendlers, 2016-06-08:
Hmmm, interseting. I was not able to reproduce the above until yesterday when I tried the tool on a raspberry PI. Maby its a colorama probem. I will investigate on this. Thanks for reporting.

MarkR42, 2016-06-09:
I'm on Ubuntu 14.04 here using the default version of colorama that appears to be installed already, and it's colorama 0.2.5.

wendlers, 2016-06-09:
Hi, 

I think I somehow figured out the problem. It is related to PySerial. In more detail:

The "repl" command in the shell uses the "Miniterm" class from PySerial. This was imported at to top of "main.py". And for some reason, on PySerial 2.x it messed with the raw input. 

Since the REPL is not (yet) supported on PySerial 2.x anyways, I pushed a new version which only imports the "Miniterm" at the moment the "repl" command is issued, and only when PySerial version is 3.x. By doing so, I was able to get the rest of the shell work also with PySerial 2.x.

Could you pleas try to confirm, that your PySerial version is also 2.x (very very likely for Ubuntu 14.x), and that the new version works? In a Python REPL you could easily check the version by:
```
import serial
serial.VERSION
```

Btw. you could try to update your Pyserial with:

```
sudo pip install --upgrade pyserial
```

If it works for you, I would close this issue, and add instead a feature request for PySerial 2.x support.

commit d0aa8bb8 (2016-06-12): Wrapper to handle PySerial 2.7 and PySerial 3.0 differences for Miniterm. Fixes #2.
closed 2016-06-12 by wendlers