wendlers/mpfshell #5: repl crashes on second start

kromosomX, 2016-06-12:
"repl" command crashes if invoked more than once on ESP8266 via sockets. I can't test via com port as on windows I can't connect via COM port

mpfs [/]> repl
'Out' object has no attribute 'buffer'
Exception ignored in: <bound method Console.__del__ of <serial.tools.miniterm.Console object at 0x02CEBC70>>
Traceback (most recent call last):
  File "C:\Python\lib\site-packages\serial\tools\miniterm.py", line 116, in __del__
    ctypes.windll.kernel32.SetConsoleOutputCP(self._saved_ocp)
AttributeError: 'Console' object has no attribute '_saved_ocp'

wendlers, 2016-06-13:
Could you please give me a hint, which python version you are using, and what is the version of PySerial. The later could be easily checked on a Python console like this:

```
import serial
serial.VERSION
```
Thanks, 
Stefan

kromosomX, 2016-06-13:
I'm using Python 3.5.1 with PySerial 3.1.1 on Windows 10. "repl" starts ok the first time and exits to mpfshell where i can run the other commands, but not "repl"

wendlers, 2016-06-13:
Hmm, 3.5 and 3.1 sounds good. I will try to reproduce on Linux, and see if I could get access to some Windows.

wendlers, 2016-06-13:
It works for me now with Python 3.5 and PySerial 3.1 on Windows. This clearly is a PySerial Windows bug, since the above described happens when creating a Miniterm instance a second time. However, by avoiding instance re-creation, it seams to do the trick for serial, websocket and telnet based connections.

closed 2016-06-13 by wendlers

kromosomX, 2016-06-14:
Both issues fixed. Thank you.

commit 82c980ea (2016-06-13): Support Windows serial port namings (COMx) which fixes #6. Also implemented a work-around which made Miniterm on Windows bomb-out when recreating an instance, this fixes #5.