wendlers/mpfshell #7: Leaving REPL kills local echo

drawkula, 2016-06-18:
Local echo is off after leaving the REPL .
I need to leave "mpfshell" and type "reset" to get my "xterm" back on trails...

Configuration:

- "webrepl" dated "Sat Jun 18 22:33:42 2016 +0200" running with Python2 on Debian8.5/AMD64.
- "MicroPython v1.8.1-77-gc7fba52" on an ESP-12E based NodeMCU Devboard clone.

wendlers, 2016-06-19:
Hi,

I think the problem is caused because of PySerial 2.7 (which is "legacy", but still used by Debian for Python 2.7). Nevertheless, this is a bug, and I will investigate on this. Until then, there are two work arounds:

1) update PySerial to version 3.x (e.g. pip install pyserial)
2) Or (recommended) if you also have Python3 on your machine, try that:

wendlers, 2016-06-19:
I tried on a Pi which is the most Debian like setup I have. I could confirm the problem you described, but I also realized that it came with an even older version of PySerial (2.6). When updating to 2.7, it worked. Could you please check which version your PySerial is? You could do so with the above command, or pull the latest version of "mpfshell" which will give a message on "repl" command if PySerial is < 2.6.

drawkula, 2016-06-19:
python-serial and python3-serial both are 2.6 on Debian-8.5

With updated mpfshell (default Python is Python2):
```
$ ./mpfshell 

** Micropython File Shell v0.7.3, sw@kaltpost.de ** 
-- Running on Python 2.7 using PySerial 2.6 --

mpfs [/]> repl

REPL needs PySerial version >= 2.7, found 2.6

mpfs [/]> _
```
Same result with `python3 ./mpfshell`

I need to think longer about updating PySerial bypassing Debian's package management with `pip`... I don't like that idea...

Instead, I've built and installed packages from python-serial sources used on Debian-9:
```
$ ls *.deb
python3-serial_3.1-2_all.deb  python-serial_3.1-2_all.deb
```
Now echo after leaving REPL works.

wendlers, 2016-06-19:
Thanks for testing that. Actually if you use

```pip install --upgrade --user pyserial```

I think it will not violate the Debian packages since it will install everything under `$HOME/.local/lib/py...`. Python itself should then use the local libs before the global ones.

drawkula, 2016-06-19:
> Python itself should then use the local libs before the global one

Other programs depending on python-serial force it being installed as deb package but Python uses my locally "pip-installed" version 3.x?

I still don't like that idea...

Let's wait and see what breaks with python-serial-3.x as deb package and and if so, I can look in Debian-9 which program needs which changes/updates to cope with the newer pyserial...

wendlers, 2016-06-19:
Yes having the package is the safer option. 

However, I will close that issue, since it does not make much sense supporting PySerial 2.6. I was playing around some more, and got the echoing problem with the REPL fixed for 2.6 but when starting the REPL a second time things started to get odd.

closed 2016-06-19 by wendlers