wendlers/mpfshell #28: mpfshell.py has to be called in its directory on W10

NoPinky, 2017-04-20:
I have mpfshell installed on my W10 machine with Python 2.7. Unlike **esptool.py**, **mpfshell.py** can't be just called from anywhere in the cmd-shell of W10. I have to `cd C:\Python27\Lib\site-packages\mp` before I can successfully call `mpfshell.py`. The reason seems to be because mpfshell keep its files under `\site-packages\mp\` and doesn't have the .py and .pyc files in the `\site-packages\` folder like esptool does
I'm new to Python, so I don't know how Python handles that kind of global calls at all.

BoKKeR, 2018-01-30:
I use this workaround 
`python -m mpfshell`
The problem is actually not with path but the setup.py as it seems to miss the executable ??? I will look into it more

skorokithakis, 2018-12-02:
This should be solved now, calling `mpfshell` from anywhere should work.

closed 2018-12-02 by skorokithakis