wendlers/mpfshell #32: Reconnecting with mpfshell to webrepl seldomly works in first try

ulno, 2017-04-27:
Connecting to webrepl via the ws option of mpfshell seems a little less reliable than using webrepl from https://github.com/micropython/webrepl.

Connecting the first time to freshly booted esp8266 usually seems to work (even if I have the feeling that it also sometimes fails when the wireless connection is weak - don't have proof for that yet). However, re-opening the websocket or even closing and re-connecting usually does not work the first time. A second re-connect often succeeds.

The webrepl web frontend seems to work nearly always (only sometimes fails when I used mpfshell in between).

Any idea why this is? What is done differently? Can this be fixed or circumvented?

A transcript of my testsession looks like this:

```
ulno#ulnoiot:$ mpfshell

** Micropython File Shell v0.8.0, sw@kaltpost.de ** 
-- Running on Python 3.5 using PySerial 3.3 --

mpfs [/]> help close
close
        Close connection to device.
        
mpfs [/]> open ws:192.168.12.57,ulnoiot

Failed to open: ws:192.168.12.57,ulnoiot

mpfs [/]> open ws:192.168.12.57,ulnoiot
Connected to esp8266
mpfs [/]> repl
>
*** Exit REPL with Ctrl+] ***

MicroPython v1.8.7-7-gb5a1a20a3 on 2017-01-09; ESP module with ESP8266
Type "help()" for more information.
>>> 
mpfs [/]> close
mpfs [/]> open ws:192.168.12.57,ulnoiot

Failed to open: ws:192.168.12.57,ulnoiot

mpfs [/]> open ws:192.168.12.57,ulnoiot
Connected to esp8266
mpfs [/]> open ws:192.168.12.57,ulnoiot

Failed to open: ws:192.168.12.57,ulnoiot

mpfs [/]> open ws:192.168.12.57,ulnoiot
Connected to esp8266
mpfs [/]> close
mpfs [/]> open ws:192.168.12.57,ulnoiot
Connected to esp8266
mpfs [/]> close
mpfs [/]> open ws:192.168.12.57,ulnoiot

Failed to open: ws:192.168.12.57,ulnoiot

mpfs [/]> open ws:192.168.12.57,ulnoiot
Connected to esp8266
mpfs [/]> close
mpfs [/]> open ws:192.168.12.57,ulnoiot

Failed to open: ws:192.168.12.57,ulnoiot

mpfs [/]> open ws:192.168.12.57,ulnoiot
Connected to esp8266
mpfs [/]> 
```

ulno, 2017-04-28:
So, I assume, the module, I need to look at is mpfshell/mp/conwebsock.py.
Would it be hard to at least enable some kind of auto-reconnect/retry?
Maybe off by default but some kind of wsretry-flag you could give on the command line.
Would it make sense to just play with the timeout? I still just don't understand why mpfshell fails so often and webrepl doesn't? Maybe the websocket handling in python is different than in javascript?
I don't see any special code handling errors in the java script part.

Micropython (https://github.com/micropython/webrepl) provides the python command-line client (webrepl-cli.py) which even does a super simple manual websocket handling (in websocket-helper.py). Maybe using that would increase stability?

Thoughts?

ulno

ulno, 2017-05-16:
I took a closer look and it seems that switching to raw repl mode messes up the micropython repl. A webrepl reset at connection close seems to fix it.

Here is my current testing fork, there is still a bit too much debug prints in there, but if you feel like testing, check it out:
https://github.com/ulno/mpfshell

skorokithakis, 2018-12-02:
Have you gotten anywhere with this? Websocket connections are currently very flaky and I'd like to fix that.