wendlers/mpfshell #33: OSX Cannot exit REPL mode using CTRL- ] combination

fstengel, 2017-05-07:
This is not quite a duplicate of #18.

Since I upgraded my python engine to Python 3.5.3 (anaconda custom (x86_64)), I cannot exit repl using ctrl-] (or ctrl-Q) anymore.

As a matter of fact, the ctrl-] shortcut was already quite a bother: the ] key cannot be accessed directly on a French keyboard. One had to switch to a US keyboard, type the sequence and then switch back to the French keyboard. All in all, a minimum of 3 key combinations (cmd-space, ctrl-], cmd-space)

Would it be possible to have ctrl-Q (or any other) as the exit key? I tried toying with the source (mpfshell.py around line 575) by replacing `chr(0x1d)` by `chr(0x11)`, but it did not change anything...

fstengel, 2017-05-08:
I went ahead and looked deeper into the problem. Since I am using anaconda, most of the packages are installed using `conda`. However a limited subset was installed using `pip`. Basically, I had two versions of `pyserial` floating around: on installed through `conda` (2.7) and one through `pip` (3.3). The `conda` one taking precedence over the `pip` one. So I had to uninstall `pyserial ` from `conda` and then uninstall/reinstall `pyserial` using `pip` and everything works as advertized.

All in all a warning: it seems the version of `pyserial` (2.7-py35_0) as installed by anaconda has a slight bug in that it prevents the proper use of ctrl-].

wendlers, 2017-08-07:
pyserial before 3.x  causes a log of pain :-/ especially the miniterm module is very buggy. I understand the problem is gone when using proper version? Thus, I close the issue.

closed 2017-08-07 by wendlers

hachpai, 2019-03-06:
I got the same issue with pyserial 3.4. I've to press ctrl-$ to exit REPL on osx mojave.