wendlers/mpfshell #34: Request: command line argument to open board directly

fstengel, 2017-05-07:
Being a lazy person :sweat_smile: I wonder if it would not be possible to use `mpfshell`as follows:
```
mpfshell --board tty.tty.SLAB_USBtoUART
```
instead of:
```
mpfshell
mpfs [/]> open tty.SLAB_USBtoUART
```
That way, one could use the shell's history to connect to the board: I tend to forget what `tty.*` my board is connected to... :flushed:

fstengel, 2017-05-08:
See relevant pull request (#35)...

ulno, 2017-05-16:
Why don't you just do: ```mpfshell -c "open tty.SLAB_USBtoUART"```

fstengel, 2017-05-16:
Indeed. However, being lazy, `mpfshell -c "open ..."` seems a bit less efficient than `mpfshell ...` Furthermore, I don't quite see the point in not opening the connection to the board as soon as one interactively uses mpfshell.

ulno, 2017-05-16:
+1 for using open anyway always as first command -> maybe specifying only one parameter should directly connect to that

Though, maybe this becomes strange when you deal with several boards in one mpfshell script.

fstengel, 2017-05-16:
Very good point. Up until now, I would have thought that to access more than one board one would have to open them using one instance of `mpfshell` per board.

wendlers, 2017-08-07:
Feature was added by push request.

closed 2017-08-07 by wendlers