wendlers/mpfshell #36: ls does not list files created on the esp8266

craftyguy, 2017-05-31:
Steps to reproduce:

1) in repl/webrepl:
```
>>> with open('app', 'w') as f:
>>>    f.write("hello")
```
2) in mpfshell:
```
mpfs [/]> ls

Remote files in '/':

```

3) in repl/webrepl:
```
>>> import os
>>> os.listdir()
['app']
```
4) in mpfshell:
```
mpfs [/]> cat app
hello
```

wendlers, 2017-08-07:
Yes, this is due to the fact, that the shell uses caching by default. For now, you could add `--nocache` when calling the shell like so:

    mpfshell --nocache

An other solution would be to invalidate the cache at the moment the we return from REPL to the shell prompt. I will flag this as an enhancement.

skorokithakis, 2018-12-02:
Fixed, thank you!

commit 195d3ed4 (2018-12-02): Clear the file cache when returning from the REPL (fixes #36)
closed 2018-12-02 by skorokithakis