wendlers/mpfshell #43: How hard to restore FUSE functionality?

sethcohn, 2017-08-14:
I realize you removed it... and looking at it, I wonder if it would still be possible to use it with WebREPL or only local ports?

wendlers, 2017-08-15:
It is not too hard, but I removed it because I don't had (and still don't have) the time to maintain it. Since the FUSE thing used the same python classes as the shell, it would in theory work with all protocols supported by the shell.

sethcohn, 2017-08-16:
I've got it working with WebREPL, it could use some improvements but it functions as expected.

Minor changes to code to get it working (I'll commit something once I'm happy with it to my fork, but if you are open to a PR, let me know.)
 (I commented out the .terminal, due to serial assumptions, had to remove the default in .commit, and change 'port' to 'constr' (since it's now doing more than just a port)
But really, relatively painless.  I was able to fuse mount an ESP8266 board onto a local directory, edit boot.py using nano, echo "boot.py" > .commit, close the mpfmount, and the altered boot.py was live on the board.  Very slick indeed.

wendlers, 2017-08-16:
sounds good. would be happy to receive a PR as far as you happy with your results.