wendlers/mpfshell #44: [macOS] error when using get.

TheTonon, 2017-08-16:
mpfs [/]> get boot.py
Traceback (most recent call last):
  File "/usr/local/bin/mpfshell", line 8, in <module>
    main()
  File "/Library/Python/2.7/site-packages/mp/mpfshell.py", line 704, in main
    mpfs.cmdloop()
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/cmd.py", line 142, in cmdloop
    stop = self.onecmd(line)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/cmd.py", line 221, in onecmd
    return func(arg)
  File "/Library/Python/2.7/site-packages/mp/mpfshell.py", line 422, in do_get
    self.fe.get(rfile_name, lfile_name)
  File "/Library/Python/2.7/site-packages/mp/retry.py", line 43, in f_retry
    return f(*args, **kwargs)
  File "/Library/Python/2.7/site-packages/mp/mpfexp.py", line 347, in get
    raise e
mp.pyboard.PyboardError: ('exception', '', 'Traceback (most recent call last):\r\n  File "<stdin>", line 3, in <module>\r\nTypeError: \'int\' object is not callable\r\n')

I've been trying to interpret and debug, but I'm not that familiar with mpfshell yet.

wendlers, 2017-08-17:
Hi,

it looks like something "unexpected" is received, and parsing of that "something" fails. Could you try the same but with logging enabled, and then attach the results from the log? Your call to `mpfshell` might look something like this with full logging:

    mpfshell --logfile mpfshell.log --loglevel DEBUG [... more args if needed ...]

TheTonon, 2017-08-17:
I am really sorry to say that I cant reproduce the issue anymore. But I guess there was something wrong with my ESP, since the last thing I did was to re-flash it to change some packages. I don't see how a package could be causing this but that is my only guess.

I'll try to roll back the firmware and my code and see if I can make the issue reproducible.