wendlers/mpfshell #53: Some filename can not put.

RainGather, 2018-04-13:
Env:
Windows 10
Anaconda 3
NodeMCU ESP8266
MicroPython latest version

I have a python file named Ultrasonic.py.
And when I do next command:

```
python -m mp.mpfshell -o COM5
md ezmpy
cd ezmpy
put Ultrasonic.py
```

There have an Error:

> Traceback (most recent call last):
>   File "C:\Anaconda3\lib\runpy.py", line 193, in _run_module_as_main
>     "__main__", mod_spec)
>   File "C:\Anaconda3\lib\runpy.py", line 85, in _run_code
>     exec(code, run_globals)
>   File "C:\Anaconda3\lib\site-packages\mp\mpfshell.py", line 711, in <module>
>     main()
>   File "C:\Anaconda3\lib\site-packages\mp\mpfshell.py", line 704, in main
>     mpfs.cmdloop()
>   File "C:\Anaconda3\lib\cmd.py", line 138, in cmdloop
>     stop = self.onecmd(line)
>   File "C:\Anaconda3\lib\cmd.py", line 217, in onecmd
>     return func(arg)
>   File "C:\Anaconda3\lib\site-packages\mp\mpfshell.py", line 369, in do_put
>     self.fe.put(lfile_name, rfile_name)
>   File "C:\Anaconda3\lib\site-packages\mp\mpfexp.py", line 537, in put
>     MpFileExplorer.put(self, src, dst)
>   File "C:\Anaconda3\lib\site-packages\mp\retry.py", line 43, in f_retry
>     return f(*args, **kwargs)
>   File "C:\Anaconda3\lib\site-packages\mp\mpfexp.py", line 302, in put
>     raise e
>   File "C:\Anaconda3\lib\site-packages\mp\mpfexp.py", line 284, in put
>     self.exec_("f = open('%s', 'wb')" % self._fqn(dst))
>   File "C:\Anaconda3\lib\site-packages\mp\pyboard.py", line 157, in exec_
>     raise PyboardError('exception', ret, ret_err)
> mp.pyboard.PyboardError: ('exception', b'', b'Traceback (most recent call last):\r\n  File "<stdin>", line 1\r\nSyntaxError: invalid syntax\r\n')

And if I rename Ultrasonic.py to u.py, and retry, it is all good, no error.
I met many times this problem, sometimes is Ultrasonic.py, sometimes is finger.py

junhuanchen, 2018-10-13:
well.
From the picture can know this error is come from the board returns information on the error.

Do not use the python -m mp. Mpfshell directly because the `lpwd` path of mpfshell is not consistent with the path of your file.

execute mpfshell directly, or check the directory returned by the `lpwd` of the board. If you're windwos, you also need to be aware that it's under the same drive and filename uppercase.

You can also try out my mpfshell. XD.

skorokithakis, 2018-12-02:
@junhuanchen Would you like to submit a PR on this (and other functionality you find useful)?