wendlers/mpfshell #54: Problems entering raw repl

fstengel, 2018-06-05:
I am currently experiencing issues when trying to connect using mpfshell (i.e. failure 19 cases in 20) with my wemos D1 mini. The error message being: "could not enter raw repl".

Browsing around I discovered that in ampy, which uses a slightly different version of pyboard.py, a delay is added before sending the double ctrl-C when entering raw repl.

So I added a 1 second delay just before line 63 in pyboard.py, in ````enter_raw_repl```` (i.e. just before ````self.con.write(b'\r\x03\x03')````). The code now looks like
````
    def enter_raw_repl(self):
        
        time.sleep(1)
        self.con.write(b'\r\x03\x03')  # ctrl-C twice: interrupt any running program

````

The result being that I can now consistently open the board. I tried reducing the delay. 0.2 second works, 0.15 doesn't. Currently I am working with a 0.5 second delay.

torntrousers, 2018-07-26:
Thanks @fstengel this has fixed the same for me too. Bit weird was that mpfshell had been working fine for me on a few different ESP8266 boards then all of a sudden it stoped working and just failing with error message: "could not enter raw repl". Nothing I tried would fix it until stumbling on solution he which fixes it.

skorokithakis, 2018-12-02:
Does this happen with USB or Websocket? Or both? Good to know that a delay fixes it, but I'd like to investigate some more before just adding a delay, it feels a bit like a hack.

torntrousers, 2018-12-02:
With USB. I know a bit more about all this now so I can go debug it this a bit more...

skorokithakis, 2018-12-02:
That would be fantastic, thank you.

torntrousers, 2018-12-06:
I've had a bit more of a look and I think its just the code runs too fast and when it does the ctrl-c's [here](https://github.com/wendlers/mpfshell/blob/master/mp/pyboard.py#L61) its just a bit too soon and the board is still reseting (the ESP8266 on the Wemos D1 mini does a reset when its connected to) so the ctrl-c's happen a bit too soon. A sleep of 0.2 seems to reliably fix it for me.

skorokithakis, 2018-12-06:
Sounds good to me, would you be interested in submitting a PR? We should probably play it safe and go for 0.5, but this sounds like an acceptable fix for me.

skorokithakis, 2018-12-06:
This should be fixed, thanks to @torntrousers.

closed 2018-12-06 by skorokithakis