wendlers/mpfshell #58: Backspace isn't working

FurmanSK, 2018-08-27:
Not sure if I'm doing something wrong but when in repl mode and try to backspace, it adds in special characters. Probably an encoding issue or something. On windows 10.

junhuanchen, 2018-09-28:
well, you can try my fork mpfshell,The author vaporized.
fix bs and up/down/left/right in windows10.

skorokithakis, 2018-12-02:
@junhuanchen I'll help maintain the project from now on, if you'd like to submit some PRs.

tsaost, 2022-07-29:
Your problem is probably due to not running mpfshell inside a console with VT100 emulation.  Since you are on Windows 10 you can follow the advise give here: https://superuser.com/questions/413073/windows-console-with-ansi-colors-handling

Or you can use ConEmu: https://github.com/Maximus5/ConEmu