wendlers/mpfshell #59: Can't connect through ws: with websocket-client version >= 0.48.0.

greblus, 2018-09-01:
Hi guys.

I've been trying to use mpfshell with my Wemos D1 mini pro and noticed that it won't connect through ws: if websocket-client version is >= 0.48.0.

In debug I see:
2018-09-01 10:51:36,860	ERROR	error from callback <bound method ConWebsock.on_message of <ConWebsock(Thread-1, started daemon 140001679963904)>>: on_message() missing 1 required positional argument: 'message'
2018-09-01 10:51:36,964	ERROR	error from callback <bound method ConWebsock.on_error of <ConWebsock(Thread-1, started daemon 140001679963904)>>: on_error() missing 1 required positional argument: 'error'
2018-09-01 10:51:37,005	ERROR	error from callback <bound method ConWebsock.on_close of <ConWebsock(Thread-1, started daemon 140001679963904)>>: on_close() missing 1 required positional argument: 'ws'
2018-09-01 10:51:41,535	ERROR

merlink01, 2018-09-27:
Same Problem here.

junhuanchen, 2018-09-28:
well, you can try my fork mpfshell,The author vaporized.

greblus, 2018-09-28:
@junhuanchen: thanks, I will try shortly.

merlink01, 2018-11-20:
I testet a little bit, seems to work better.

Will there be a pull request, and someday a new pipy version???

skorokithakis, 2018-12-02:
I will accept a PR for this.

xnorbt, 2019-11-26:
Removing the second parameter 'ws' in on_message and on_error in conwebsock.py seems to fix the issue (at least it is working for me). I.e.:
change `def on_message(self, ws, message):` to `def on_message(self, message):`
Same with on_error.

skorokithakis, 2019-11-26:
Fantastic, can you issue a PR that updates the dependency version and removes the parameter?

xnorbt, 2019-11-26:
Which version should I put in the requirements file? I have tested it with websocket_client-0.56.0.