wendlers/mpfshell #60: Not usable anymore on Circuitpython

Tasm-Devil, 2018-10-11:
It is not your fault, but it seams that adafruit made some changes in their circuitpython code. 
See here: https://github.com/adafruit/circuitpython/issues/1029
I get the same error when i try to connect to a circuitpython board.
Could u fix this, plz

junhuanchen, 2018-10-13:
yep.

stlehmann, 2018-12-28:
We could tackle this by trying to import the modules *ubinascci* and *uos* and if this fails we go for *binascci* and *os*. Another idea would be to check if it is micropython or circuitpython and change the imports accodingly. I'll give it a try.

skorokithakis, 2018-12-28:
The former sounds like a good solution, try to import the first two and if there's an `ImportError`, fall back to the other two.

stlehmann, 2018-12-28:
> I've pushed a change that tries os as uos, can you see if that works and let me know?
Yes, this seems to work now. 

But there is another problem with binascii. I just discovered that circuitpython does not provide either a binascii or a ubinascii module.

skorokithakis, 2018-12-28:
Ah, that seems rather harder to fix. The solution for this is the same as in #69, which is to use binary mode to transfer files.

stlehmann, 2018-12-28:
Adafruit makes the following statement in their [Circuit Python documentation](https://circuitpython.readthedocs.io/en/3.x/docs/library/index.html#python-standard-libraries-and-micro-libraries): 

> Some libraries are only enabled only WiFi-capable ports (ESP8266, nRF) because they are typically used for network software: binascii, hashlib, uheapq, uselect, ussl. Not all of these are enabled on all WiFi-capable ports.

That explain why mpfshell doesn't work on the Feather M0. Apparently one would need to enable the binascii module in the firmware before uploading it.

closed 2021-08-14 by Tasm-Devil