wendlers/mpfshell #62: Use uos.uname instead of os.uname

stlehmann, 2018-11-18:
mfpshell uses the `uname` function to get information about the board that it connects to. In the [Micropython documentation](https://docs.micropython.org/en/latest/library/uos.html) this function lives in the "uos" module. However mpfshell uses `os.uname()` which is not a problem as long as one does not install the "os" module from micropython-lib which offers advanced functionality. This os module does not support `uname` as it is meant to lean on to CPythons `os` module.

A proper solution is to use `uos.uname` in the mpshell code. I fixed this in https://github.com/wendlers/mpfshell/pull/51.

skorokithakis, 2018-12-02:
Your fix in #51 doesn't seem like it'll work, since the module gets imported as `os` and it's called as `os` everywhere else. Have you tested it?

stlehmann, 2018-12-02:
Yes, I tested it on an ESP32 and it works just fine because uos gets loaded automatically on startup. You are right, the import for the `uos` module is missing. We should add this to the existing imports even the `uos` module gets imported automatically on startup. Using `os` everywhere else is also OK because the called functions exist in `uos` and `os`. However with `uname` that is not the case so `uos.uname` needs to be used here.

skorokithakis, 2018-12-02:
We should be consistent and use the same module name everywhere, so we should migrate everything from `os` to `uos`. You're right that it gets loaded on startup, but I still prefer the explicit import. I would merge a PR that changes all `os.` to `uos.` calls, and also imports `uos` at the start. I'll implement it if I have time, or if you want to take a stab at it sooner, even better!

skorokithakis, 2018-12-02:
This should be fixed in 3af4, can you test>?

closed 2018-12-02 by skorokithakis