wendlers/mpfshell #69: Performance issue in put/get

kif, 2018-12-19:
Hi,

I am facing many problems to upload/download (large) files with mpfshell, while it looks better with the ampy tool from adafruit. 
I have the feeling mpfshell transfers all data in ascii while ampy does this binary but I am not completely sure about that.
Can you confirm ? how difficult would it be to  offer another transfer mode ?

Cheers,
Jerome

skorokithakis, 2018-12-19:
Hello,  
you are correct: https://github.com/wendlers/mpfshell/blob/master/mp/mpfexp.py#L268

I wonder what it would take to switch to binary mode, transfer and switch back. I'll have to look into how ampy does it.

As always, PRs welcome!

skorokithakis, 2018-12-19:
After some investigation, it seems that mpfshell already includes functions like `read_until`, `enter_raw_repl`, etc, but they aren't used. They should be.

skorokithakis, 2018-12-19:
For reference, this is how ampy does it: https://github.com/adafruit/ampy/blob/master/ampy/files.py#L204