wendlers/mpfshell #71: Files in subdirectories inaccessible if running mpfshell from Windows

zouden, 2019-02-06:
On micropython systems with the /flash subdirectory, all get/put commands fail when running mpfshell from Windows. This is caused by the use of `os.path.join` in mpfexp.py which results in file paths joined with \\\\ instead of /. Micropython's file system uses /.

I managed to fix it by `import posixpath` in mpfexp.py followed by replacing all instances of `os.path.join` with `posixpath.join`.

skorokithakis, 2019-02-06:
Would you care to issue a PR? This shouldn't break existing functionality, right?

skorokithakis, 2019-02-06:
Hmm, is that module in the standard library?

zouden, 2019-02-06:
Yep, I'll do a PR. 

It's in the standard library, but I'm not sure if it's the 'best' way of forcing posix-like paths. Some documentation here:
https://docs.python.org/3/library/pathlib.html

PR #72 (2019-02-07): Fix posix-style paths from Windows
This fixes #71 on my Windows system and should work fine on Linux/MacOS but has not been tested.
closed 2019-02-07 by skorokithakis