wendlers/mpfshell #75: How to run scripts

cecchisandrone, 2019-03-27:
Is it possible to run scripts using mpfshell (like run command in ampy)? 
Thanks.

barbudor, 2019-03-27:
I don't know ampy but you can use the REPL console (>>>) if this is what you are looking for.

skorokithakis, 2019-03-27:
That is correct, you can use the REPL and run whatever script is on the microcontroller.

closed 2019-03-27 by skorokithakis

cecchisandrone, 2019-03-27:
> I don't know ampy but you can use the REPL console (>>>) if this is what you are looking for.

What is the command?

barbudor, 2019-03-28:
Have you tried to use `?` or`help`commands ?
It's there, hidden in plain sight:
[image]

cecchisandrone, 2019-03-28:
I mean running a script from repl

barbudor, 2019-03-28:
Oh. Sorry for the misunderstanding and little moking.
That's not related to mpfshell itself. It would be the same whatever terminal you use (uPyCraft, Mu, ...)
I'm using `import  nameofscript`  (without the .py) but not sure if that's the only way.
Note that `import` is not just exactly 'running'. That's importing and compiling the code and executing the part of the code that is at the root level (outside of any def or class). Which means that you cannot re-import again to re-execute as the module has already been imported.
I suggest you move this question/discussion to MicroPython forum: https://forum.micropython.org/

cecchisandrone, 2019-03-28:
Normally I do this in ampy (run command). The problem of that shell is that you see the output of the script only when it terminates, so I was trying to find an alternative.

skorokithakis, 2019-03-28:
Hmm, I'm not entirely sure what you want to do. Do you want to run a script that's on the computer, or on the MPU?

cecchisandrone, 2019-03-29:
On the MPU.