wendlers/mpfshell #76: Hangs after (successful?) entering repl

supersjimmie, 2019-03-28:
I have an esp32 with loboris micropython.
When I completely clear the esp32 and put a fresh flashed image, I can enter mpfshell, access the files and all seems fine. But when I want to enter repl, the system completely hangs.

```
me:~/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD$ ./BUILD.sh -v -p /dev/ttyS12 erase

----------------------------------------
MicroPython for ESP32 with psRAM support
----------------------------------------

======================
Erasing ESP32 Flash...
======================
Erasing entire flash...
esptool.py v2.5.0
Serial port /dev/ttyS12
Connecting.....
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse
MAC: 80:7d:3a:b9:9b:6c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Erasing flash (this may take a while)...
Chip erase completed successfully in 5.3s
Hard resetting via RTS pin...
OK.

me:~/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD$ ./BUILD.sh -v -p /dev/ttyS12 flash

----------------------------------------
MicroPython for ESP32 with psRAM support
----------------------------------------

=========================================
Flashing MicroPython firmware to ESP32...
=========================================
Building partitions from /home/me/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/partitions_mpy.csv...
Flashing binaries to serial port /dev/ttyS12 (app at offset 0x10000 )...
esptool.py v2.5.0
Serial port /dev/ttyS12
Connecting....
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse
MAC: 80:7d:3a:b9:9b:6c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 19440 bytes to 11566...
Wrote 19440 bytes (11566 compressed) at 0x00001000 in 0.2 seconds (effective 938.3 kbit/s)...
Hash of data verified.
Compressed 144 bytes to 69...
Wrote 144 bytes (69 compressed) at 0x0000f000 in 0.0 seconds (effective 202.2 kbit/s)...
Hash of data verified.
Compressed 1956560 bytes to 1130961...
Wrote 1956560 bytes (1130961 compressed) at 0x00010000 in 17.4 seconds (effective 900.7 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.0 seconds (effective 4069.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
OK.

me:~/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD$ mpfshell -c "open ttyS12"
Connected to esp32_LoBo

** Micropython File Shell v0.9.0, sw@kaltpost.de **
-- Running on Python 2.7 using PySerial 3.4 --

mpfs [/flash]> ls

Remote files in '/flash':

 <dir> ..
       boot.py

mpfs [/flash]> cat boot.py
# This file is executed on every boot (including wake-boot from deepsleep)
import sys
sys.path[1] = '/flash/lib'

mpfs [/flash]> repl
>
*** Exit REPL with Ctrl+] ***

MicroPython ESP32_LoBo_v3.2.24 - 2018-09-06 on ESP32 board with ESP32
Type "help()" for more information.
>>>

```
Here nothing works anymore, I have to kill the mpfshell to get out.

skorokithakis, 2019-03-28:
Hmm, this sounds like something isn't initialized properly. Unfortunately, I don't have an ESP32 to test this on, if you could debug I would appreciate it.

supersjimmie, 2019-03-29:
@skorokithakis  how/what can I debug to help solve this?
(I can access repl via normal serial like minicom or putty)

supersjimmie, 2019-03-29:
Funny, when I use:
``` mpfshell -c "open ttyS12" --nocolor --nocache --logfile ~/mpfshe
ll.log --loglevel "DEBUG"```
It works normal.
Then go back without all the options the repl hangs.

supersjimmie, 2019-03-29:
Or even more fun, only adding the log makes it work!
```
mpfshell -c "open ttyS12" --logfile ~/mpfshell.log
```

supersjimmie, 2019-03-29:
seemed to work when I just added the log with debug level,  but then after going in and out the repl it hung again.
the log doesn't show much specific.

```
cat ~/mpfshell.log
2019-03-29 09:29:10,632 INFO    Micropython File Shell v0.9.0 started
2019-03-29 09:29:10,632 INFO    Running on Python 2.7 using PySerial 3.4
19-03-29 09:29:10,766 DEBUG   serial write >
019-03-29 09:29:10,766 DEBUG   serial write >
2019-03-29 09:29:10,770 DEBUG   serial read < r
2019-03-29 09:29:10,770 DEBUG   serial read < a
2019-03-29 09:29:10,770 DEBUG   serial read < w
2019-03-29 09:29:10,770 DEBUG   serial read <
2019-03-29 09:29:10,770 DEBUG   serial read < R
2019-03-29 09:29:10,770 DEBUG   serial read < E
2019-03-29 09:29:10,770 DEBUG   serial read < P
2019-03-29 09:29:10,771 DEBUG   serial read < L
2019-03-29 09:29:10,773 DEBUG   serial read < ;
2019-03-29 09:29:10,773 DEBUG   serial read <
2019-03-29 09:29:10,773 DEBUG   serial read < C
2019-03-29 09:29:10,773 DEBUG   serial read < T
2019-03-29 09:29:10,774 DEBUG   serial read < R
2019-03-29 09:29:10,774 DEBUG   serial read < L
2019-03-29 09:29:10,774 DEBUG   serial read < -
2019-03-29 09:29:10,774 DEBUG   serial read < B
2019-03-29 09:29:10,774 DEBUG   serial read <
2019-03-29 09:29:10,774 DEBUG   serial read < t
2019-03-29 09:29:10,774 DEBUG   serial read < o
2019-03-29 09:29:10,774 DEBUG   serial read <
2019-03-29 09:29:10,774 DEBUG   serial read < e
2019-03-29 09:29:10,774 DEBUG   serial read < x
2019-03-29 09:29:10,774 DEBUG   serial read < i
2019-03-29 09:29:10,774 DEBUG   serial read < t
2019-03-29 09:29:10,775 DEBUG   serial read <
2019-03-29 09:29:10,775 DEBUG   serial read <

2019-03-29 09:29:10,775 DEBUG   serial read < >
2019-03-29 09:29:10,775 DEBUG   serial write > import uos, sys
2019-03-29 09:29:10,785 DEBUG   serial write > 
2019-03-29 09:29:10,788 DEBUG   serial read < OK
2019-03-29 09:29:10,796 DEBUG   serial read < 
2019-03-29 09:29:10,796 DEBUG   serial read < 
2019-03-29 09:29:10,797 DEBUG   serial read < >
2019-03-29 09:29:10,797 DEBUG   serial write > try:
    import ubinascii
except ImportError:
    import binascii as ubinascii
2019-03-29 09:29:10,808 DEBUG   serial write > 
2019-03-29 09:29:10,810 DEBUG   serial read < OK
2019-03-29 09:29:10,820 DEBUG   serial read < 
2019-03-29 09:29:10,821 DEBUG   serial read < 
2019-03-29 09:29:10,821 DEBUG   serial read < >
2019-03-29 09:29:10,821 DEBUG   serial write > print(uos.getcwd())
2019-03-29 09:29:10,832 DEBUG   serial write > 
2019-03-29 09:29:10,833 DEBUG   serial read < OK
2019-03-29 09:29:10,847 DEBUG   serial read < /
2019-03-29 09:29:10,848 DEBUG   serial read < f
2019-03-29 09:29:10,849 DEBUG   serial read < l
2019-03-29 09:29:10,849 DEBUG   serial read < a
2019-03-29 09:29:10,850 DEBUG   serial read < s
2019-03-29 09:29:10,850 DEBUG   serial read < h
2019-03-29 09:29:10,850 DEBUG   serial read <
2019-03-29 09:29:10,851 DEBUG   serial read <

2019-03-29 09:29:10,851 DEBUG   serial read < 
2019-03-29 09:29:10,851 DEBUG   serial read < 
2019-03-29 09:29:10,852 DEBUG   serial read < >
2019-03-29 09:29:10,852 DEBUG   serial write > print(uos.uname()[0])
2019-03-29 09:29:10,863 DEBUG   serial write > 
2019-03-29 09:29:10,865 DEBUG   serial read < OK
[... 318 lines cut ...]
2019-03-29 09:35:40,492 DEBUG   serial read < e
2019-03-29 09:35:40,493 DEBUG   serial read < s
2019-03-29 09:35:40,493 DEBUG   serial read < p
2019-03-29 09:35:40,494 DEBUG   serial read < 3
2019-03-29 09:35:40,494 DEBUG   serial read < 2
2019-03-29 09:35:40,494 DEBUG   serial read < _
2019-03-29 09:35:40,494 DEBUG   serial read < L
2019-03-29 09:35:40,495 DEBUG   serial read < o
2019-03-29 09:35:40,495 DEBUG   serial read < B
2019-03-29 09:35:40,495 DEBUG   serial read < o
2019-03-29 09:35:40,495 DEBUG   serial read <
2019-03-29 09:35:40,496 DEBUG   serial read <

2019-03-29 09:35:40,496 DEBUG   serial read < 
2019-03-29 09:35:40,496 DEBUG   serial read < 
019-03-29 09:35:42,586 DEBUG   serial write >
2019-03-29 09:35:42,587 DEBUG   serial read < >
2019-03-29 09:35:42,598 DEBUG   serial read <
2019-03-29 09:35:42,598 DEBUG   serial read <
MicroPython ESP32_LoBo_v3.2.24 - 2018-09-06 on ESP32 board with ESP32
Type "help()" for more information.
>>>

```
So nothing at the moment it hangs.

janoist1, 2020-07-29:
@supersjimmie  how to debug ?