wendlers/mpfshell #78: Impossible to exit REPL on some keyboard layouts

powermosfet, 2019-05-11:
On certain keyboard layouts, you need a modifier key to type `]`, and this makes the REPL not recognize it.

Specifically, I use a nordic layout, and I'm on Linux.

Perhaps there should be an option to enable the Windows-variant, `Ctlr-Q`

I have to either modify the source code so mpfshell thinks I'm on windows (I have done this and it seems to work), or close the whole terminal when I want to get out of the REPL

skorokithakis, 2019-05-12:
Hmm, which modifier key do you need to press?

powermosfet, 2019-05-13:
To type a `]` i need to hold down `AltGr` and press `9`, so `Ctrl+]` kind of becomes `Ctrl+AltGr+9` but that does not work. 

I have the  changed to `self.repl.exit_character = chr(0x03)` locally so I can exit with `Ctrl+C`

skorokithakis, 2019-05-16:
Oh I see. Sure, if it won't conflict with anything else, an alias would be nice. Do you have any time to work on this? If not, I can do it but it might take a while.

powermosfet, 2019-05-20:
I can make a PR, sure. 

I think the cleanest option would be to have a command line flag that force it to to use `Ctrl+Q`, even on non-windows systems. Do you agree?

```
$ mpfshell -o ttyUSB0 -q

$ mpfshell -o ttyUSB0 --ctrl-q
```

skorokithakis, 2019-05-20:
That sounds good, although I'm wondering if there's value in making it settable (`--exit-key=q`), but it should probably work your way if that's a standard key combo.

Also, it might not even be an option but just a default key so both work, since this combination is currently unused, AFAIK.

powermosfet, 2019-05-25:
Ok, this is really weird.

I set out to make a pull request for this today, and when I tested it, the `Ctrl+]` combination... worked. So yeah. I'm a bit stumped. I don't know why it didn't work before or what has changed since then. 

So I guess this has become non-issue. :laughing:

skorokithakis, 2019-05-25:
Oh I hate those kinds of problems :P I've had my fair share of reported bugs that worked when I tried them again. Let me know if you hit this problem again and manage to reproduce it!

closed 2019-05-25 by skorokithakis