wendlers/mpfshell #81: Error when trying to get into python shell (repl)

bartvollebregt, 2019-07-31:
I get an error when trying to get into the python shell by using `repl`

```
mpfs [/]> repl
Traceback (most recent call last):
  File "/usr/local/bin/mpfshell", line 11, in <module>
    load_entry_point('mpfshell==0.9.1', 'console_scripts', 'mpfshell')()
  File "build/bdist.linux-x86_64/egg/mp/mpfshell.py", line 819, in main
  File "/usr/lib/python2.7/cmd.py", line 142, in cmdloop
    stop = self.onecmd(line)
  File "/usr/lib/python2.7/cmd.py", line 221, in onecmd
    return func(arg)
  File "build/bdist.linux-x86_64/egg/mp/mpfshell.py", line 600, in do_repl
  File "build/bdist.linux-x86_64/egg/mp/term.py", line 75, in <module>
ImportError: cannot import name unichr
```

skorokithakis, 2019-07-31:
That directory looks weird, how did you install it?

bartvollebregt, 2019-08-01:
I installed it with `sudo pip install mpfshell`
But `/usr/local/bin/mpfshell` isn't such a weird location is it?

Isn't `build/bdist.linux-x86_64/egg/mp/mpfshell.py` the location on the micro controller?

skorokithakis, 2019-08-01:
No, that code runs on the PC, not the microcontroller, that's why I'm confused. I don't think it should be under `build/`. Are you running it from a place where you have a checkout of mpfshell?

elishaROBINSON, 2019-08-10:
facing the same issue
im running mpfshell in ubuntu 18.04
as root
im trying to connect to a esp8266 with micropython on it 
should i reinstall micropython on the esp8266

bartvollebregt, 2019-08-12:
Hmmm I installed it using:

``` 
git clone https://github.com/wendlers/mpfshell .
sudo pip install -r requirements.txt
sudo python setup.py install
```

This is my install location:
``` 
➜  ~ which mpfshell
/usr/local/bin/mpfshell
```

I am running disco dingo:
```
➜  ~ cat /etc/*-release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=19.04
DISTRIB_CODENAME=disco
DISTRIB_DESCRIPTION="Ubuntu 19.04"
NAME="Ubuntu"
VERSION="19.04 (Disco Dingo)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 19.04"
VERSION_ID="19.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=disco
UBUNTU_CODENAME=disco
````

skorokithakis, 2019-08-12:
Hmm, I can't reproduce this (lots of intermediate output elided to avoid spam):

```
02:48:15 $ git clone git@github.com:wendlers/mpfshell.git
Resolving deltas: 100% (440/440), done.

02:48:45 $ virtualenv --python /usr/bin/python3 /tmp/venv
Installing setuptools, pkg_resources, pip, wheel...done.

stavros@zeus:/t/mpfshell (master|✔) 
02:48:49 $ . /tmp/venv/bin/activate.fish

(venv) stavros@zeus:/t/mpfshell (master|✔) 
02:48:51 $ pip install -Ur requirements.txt
Finished processing dependencies for mpfshell==0.9.1

(venv) stavros@zeus:/t/mpfshell (master|✔) 
02:48:58 $ mpfshell -o ttyUSB0
Connected to esp8266

** Micropython File Shell v0.9.1, sw@kaltpost.de ** 
-- Running on Python 3.7 using PySerial 3.4 --

mpfs [/]> repl
>
*** Exit REPL with Ctrl+] ***

MicroPython v1.11-8-g48dcbbe60 on 2019-05-29; ESP module with ESP8266
Type "help()" for more information.
>>> 
```

bartvollebregt, 2019-08-14:
Can you try with python2 in an non-virtual env?

jcw, 2019-12-24:
I _think_ I'm running into the same issue (on Ubuntu 19.10):

```
$ mpfshell ttyACM0
Connected to pyboard

** Micropython File Shell v0.9.1, sw@kaltpost.de **
-- Running on Python 2.7 using PySerial 3.4 --

mpfs [/flash]> ls

Remote files in '/flash':

 <dir> ..
       README.txt
       boot.py
       main.py
       pybcdc.inf

mpfs [/flash]> repl
Traceback (most recent call last):
  File "/home/jcw/.local/bin/mpfshell", line 10, in <module>
    sys.exit(main())
  File "/home/jcw/.local/lib/python2.7/site-packages/mp/mpfshell.py", line 819, in main
    mpfs.cmdloop()
  File "/usr/lib/python2.7/cmd.py", line 142, in cmdloop
    stop = self.onecmd(line)
  File "/usr/lib/python2.7/cmd.py", line 221, in onecmd
    return func(arg)
  File "/home/jcw/.local/lib/python2.7/site-packages/mp/mpfshell.py", line 600, in do_repl
    from mp.term import Term
  File "/home/jcw/.local/lib/python2.7/site-packages/mp/term.py", line 75, in <module>
    from serial.tools.miniterm import Miniterm, ConsoleBase, unichr
ImportError: cannot import name unichr
```

jcw, 2019-12-24:
To follow up on myself (with apologies): when python3 is the default, the REPL in `mpfshell` works. This can be accomplished with `sudo update-alternatives --install /usr/bin/python python ...` (I switched from the default from 2.7.17 to 3.7.5).