wendlers/mpfshell #95: Can not transfer binary files after upgrade to macOS Big Sur

iBobik, 2020-12-06:
With the same ESP32 device, board and firmware based on MicroPython 1.13 I've used to copy the same text and binary files on previos macOS and it worked. When I upgraded to BIg Sur only binary fiels stopped working.

Tried to reinstall and upgrade mpfshell by pip3 and brew.

skorokithakis, 2020-12-06:
Hmm, unfortunately I don't have a Mac to test on, so any PRs would be appreciated.

yaleman, 2021-01-03:
@iBobik I was able to test this on a lolin32 with Big Sur on an M1 mac... what versions of things are you running?

```
$ mpfshell -o /tty.usbserial-2340
Connected to esp32

** Micropython File Shell v0.9.1, sw@kaltpost.de **
-- Running on Python 3.9 using PySerial 3.5 --
```

iBobik, 2021-01-03:
Hello,
the same:

~ $ mpfshell -o cu.usbserial-1420

Connected to esp32

** Micropython File Shell v0.9.1, sw@kaltpost.de **

-- Running on Python 3.9 using PySerial 3.5 --

Honza Pobořil

skorokithakis, 2021-02-17:
Closing as a duplicate of #97.

closed 2021-02-17 by skorokithakis