wendlers/mpfshell #96: mpfshell does not connect to device on MacOs BigSur

AltamarMx, 2021-01-16:
Hi, 
i have been using mpfshell for over two years, mainly in linux, but now in MacOs.

I can flash the firmware to ESP8266/32 devices and access to them from linux, but if i switch to MacOs Big Sur  i get this after 15 seconds aprox:

m

> pfs [/]> open tty.SLAB_USBtoUART
> b'\x1b[0;32mI (539) cpu_start: Pro cpu up.\x1b[0m\r\n\x1b[0;32mI (539) cpu_start: Application information:\x1b[0m\r\n\x1b[0;32mI (539) cpu_start: Compile time:     Sep  2 2020 03:00:08\x1b[0m\r\n\x1b[0;32mI (542) cpu_start: ELF file SHA256:  0000000000000000...\x1b[0m\r\n\x1b[0;32mI (548) cpu_start: ESP-IDF:          v3.3.2\x1b[0m\r\n\x1b[0;32mI (553) cpu_start: Starting app cpu, entry point is 0x40082f30\x1b[0m\r\n\x1b[0;32mI (0) cpu_start: App cpu up.\x1b[0m\r\n\x1b[0;32mI (563) heap_init: Initializing. RAM available for dynamic allocation:\x1b[0m\r\n\x1b[0;32mI (570) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM\x1b[0m\r\n\x1b[0;32mI (576) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM\x1b[0m\r\n\x1b[0;32mI (582) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM\x1b[0m\r\n\x1b[0;32mI (589) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM\x1b[0m\r\n\x1b[0;32mI (595) heap_init: At 3FFCA9E8 len 00015618 (85 KiB): DRAM\x1b[0m\r\n\x1b[0;32mI (601) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM\x1b[0m\r\n\x1b[0;32mI (607) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM\x1b[0m\r\n\x1b[0;32mI (614) heap_init: At 4009DE28 len 000021D8 (8 KiB): IRAM\x1b[0m\r\n\x1b[0;32mI (620) cpu_start: Pro cpu start user code\x1b[0m\r\n\x1b[0;32mI (303) cpu_start: Starting scheduler on PRO CPU.\x1b[0m\r\n\x1b[0;32mI (0) cpu_start: Starting scheduler on APP CPU.\x1b[0m\r\nMicroPython v1.13 on 2020-09-02; ESP32 module with ESP32\r\nType "help()" for more information.\r\n>>> '
> 
> could not enter raw repl

The same device can be accesed from a virtual linux machine in the same computer, which makes me think it's a problem with mpfshell.

Also, i am able to connect to REPL using screen.

I am not very sure when this problem started, most of the time i am a linux user, but not for now, since i am working from home with my macbook.

I have tried installing mpfshell from pip3, clonning the repo and installing, with no sucess.

AltamarMx, 2021-02-05:
If i roll back to v0.9.0
mpfshell works in MacOs

skorokithakis, 2021-02-17:
This seems like a duplicate of #97, please help debug/file a PR if you can.

closed 2021-02-17 by skorokithakis