wendlers/mpfshell #97: Mpfshel closes when connecting to esp82 or esp32

RAMZUG, 2021-02-04:
Esptool and mpfshell were installed on linux mint.

The firmware is deleted and correctly loaded to esp82 and esp32, python 3.8 and pyserial 3.5 are used. When trying to connect to esp82 or esp32 using mpfshell, the connection is established, but the mpfshell closes. The above is repeated regardless of the ESPxx that is used. The firmware has been reinstalled and the card changed (I have tested it on 8 different cards)

Any idea what could be wrong?
[image]

AltamarMx, 2021-02-05:
This only happens with version 0.9.2,
if you roll back to 0.9.1 works well.

skorokithakis, 2021-02-05:
That's interesting, can someone bisect and find the offending commit?

edcline, 2021-02-10:
same problem with 0.9.2 on windows -> esp32
this is my first install - how do you do a 'roll back' to 0.9.1.   I have googled it but it is not clear and I am new to python/pip...
Thank You

AltamarMx, 2021-02-10:
uninstall first
pip uninstall mpfshell
reinstall specifying version
pip install mpfshell==0.9.1
pip or pip3, whetever you are using.

commit 4cfcf4f8 (2021-05-05): Fix #97 (#101)

* typo

* Revert "fix: Remove second argument from on_error and on_message in conwebsock.py to make ws connections work again (#87)"

This reverts commit 378d24d2b85d0dca856aec3600d1dc176a6738b0.

* use uos

* os is not imported yet!
closed 2021-05-05 by skorokithakis