wendlers/mpfshell #103: version 0.9.2 not work

kenjican, 2021-07-27:
execute mpfshell, open ttyUSB0 , it shows connected and then exit mpfshell. I used to upload files with version 0.9.1 ,it works well.

OS: LinuxMint 20.2
Python 3.8
Pyseiral 3.5

ianrepsch, 2021-08-04:
Have the same problem, works fine in popOs 20.04 (Pyserial 3.4, Python 3.8) with 0.9.1.
But didn't work on Arch Linux machine (Pyserial 3.5, Python 3.9.6).
pip3 list shows 0.9.2, uninstalled that then 'sudo pip3 install mpfshell==0.9.1'
Now works fine on Arch (I didn't try 0.9.2 on popOS to see if it breaks).

formatc1702, 2021-10-02:
I have the same issue with 0.9.2. Falling back to 0.9.1 works.

MacOS 11.6
Python 3.9

hyx0329, 2022-07-25:
I found the reason. `cmd.Cmd.cmdloop` uses return value from `do_*` as a flag for stop([source code here](https://github.com/python/cpython/blob/94eb1e9789eacf855919cddade7d7d0d26b4f08d/Lib/cmd.py#L217)), which cause the loop to exit if True.

Before https://github.com/wendlers/mpfshell/commit/190bc7557d7c8bf72947c1fff212301962fe031d, the return value was `None` and after that, successfully opening a connection returns `True` which causes the `cmdloop` to exit. I'd suggest to use a better approach to expose the connection status.

skorokithakis, 2022-07-25:
Thank you, @hyx0329! Would you be interested in submitting a PR?

PR #107 (2022-07-26): Prevent shell exit after a successful connection
This pr fix issue introduced by 190bc7557d7c8bf72947c1fff212301962fe031d, close #103.
commit b4e47ffa (2022-07-26): fix: Prevent cmd exiting after a successful connection (#107)

This commit fix #103 which causes the command line to exit after successfully
opening a connection.
closed 2022-07-26 by skorokithakis