wendlers/mpfshell #105: couldn't connect via websocket

pontius-g, 2022-06-05:
Hi

Trying to open websocket connection to webrepl, got the following error:
`TypeError: ConWebsock.on_message() takes 2 positional arguments but 3 were given`

it appeared, because there is a websocket object passed first to actual message:
`(<websocket._app.WebSocketApp object at 0x000002C0B5ADA260>, 'OK')`

the following correction of original function .on_mesage() in mp/conwebsock.py at line 75, fixed its behavior:
```
def on_message(self, *message):
        self.fifo.extend(message[-1])
```

skorokithakis, 2022-06-05:
Thanks for this! Could you open a PR?

pontius-g, 2022-06-05:
I could) but I see, there is no need. 
It was fixed in other way at main branch already.
It seams, **pip** provides an old version and that is why got this issue.

Thanks

closed 2022-06-05 by pontius-g

skorokithakis, 2022-06-05:
Ah, I'll have to make a new release then.